ADTPro/adtpro #185: Win 11 connection

GaryM52, 2023-12-29:
How do I connect an Apple IIc with the Imagewriter cable (DB25) to the USB port of my Windows 11 PC?  I have a Microtech International USB XpressSCSI cable but no drivers.  Last available driver works under Windows 98.  Any help will be appreciated.

david-schmidt, 2023-12-30:
If you don't have the right cables, I'd suggest picking up what you need from [RetroFloppy](https://retrofloppy.com/products/).

GaryM52, 2024-01-05:
Thanks for the suggestion.  I reread the web-page and realized I needed a null modem cable, not the Imagewriter I cable.  I have been able to boot the Apple IIc with ADTPro.  Thanks again

thac202, 2024-08-07:
Hello all,
I have the following message "No serial port or RXTX library not installed" 
How to solve the problem ?
Thank you

david-schmidt, 2024-08-07:
The main part of the server program isn't able to find the serial library as it expects.  How are you starting your server (which leads to other questions like... what operating system are you on, what does your startup script look like, where is your server located in the filesytem, ...)?

thac202, 2024-08-07:
I find the problem, I was connected via an USB hub. Now it's ok. A new problem is appearing : How to configure the IIc ?
I boot ADTPRO 2.10 onto the IIc and I have this message "waiting for host reply, ESC CANCEL"
On the PC side I have the server is connected 115200 bauds. I want to send disk image to the IIc, what is the procedure ?

david-schmidt, 2024-08-07:
This is (another) cabling/connection problem.  To get help, you'd need to describe very accurately how your host machine is connected physically to your IIc including the type of null modem cable you have going from the IIc to the USB RS232 adapter.  If it happens that you're on Windows, you would need to have selected the correct COM port on the server side to be sure you're talking to the port you think you're talking to.  All this is detailed on the serial connections page:
https://adtpro.com/connectionsserial.html#DIN5
https://adtpro.com/connectionsserial.html#USB_cabling

thac202, 2024-08-07:
Ok it works fine !
I use the procedure "from bare metal" and I create a new  ADTPRO 2.10 disk on the IIc side. 
Thank you for your time !

david-schmidt, 2024-08-07:
Excellent - glad that worked for you!

closed 2024-08-07 by david-schmidt