ADTPro/adtpro #186: Batch send crashes

GaryM52, 2024-01-07:
I'm using ADTPro 2.1.0, and have bootstrapped an Apple IIc from a Win 11 PC.  Everything worked fine the first day, but when I came back the next day to do a batch transfer the following happened:  The process was normal up to Sending the first 2 blocks of data which were not received by the Win 11, the IIc screen then filled with garbage (image attached), and the external disk started a continuous spin (light on).  I then disconnected the link from the Server (Win 11), hit several ESCs and Returns on the Client (IIc) and the disk stopped.
I thought the Anti-Virus on the Server might be the problem, disabled anything related to the file transfer, and tried again with the same result.  Any help will be appreciated.  Thanks
[image]

david-schmidt, 2024-01-09:
it sounds like there's something about the disk you're transferring that is getting things confused... if you send that disk just normally, does it crash too?  I.e. my working theory is it has nothing to do with batch-ness, but is instead due to the disk itself.  Can you narrow things down by checking if you can send it successfully singly, and/or do other disks work when sending in batch?

GaryM52, 2024-01-09:
This first occurred when I tried to send the Appleworks Boot Disk over to the server, and have tried Send as well as batch of Data Disks with the same result each time.  I even tried to bootstrap the program again with no success.  The client software seems to be running fine on the initial disks (I made two copies and they both have the same result), but I did notice an old filename for the external disk after I had swapped it, like ADTPro didn't notice the change.

GaryM52, 2024-01-09:
I have checked the disks and I may have tried to send a faulty Appleworks disk.  It didn't boot properly, and I noted it was write--protected.  Some of the data disks I tried later had some I/O errors for specific files, but some were readable with Appleworks.  Afterwards I tried to Send a good data disk to the server, and had a similar failure to the one noted above.  ADTPro reads the blocks of data, then tries to 'send' the blocks, and the failure occurs.  There is no activity on the server, but it is ready to receive the file.  I terminate by Disconnecting the server after the "Disk busy" status on the client.

david-schmidt, 2024-01-10:
Again, is there any disk that sends successfully (either in batch or singly), or is it not sending anything at all successfully?

Can you run the IIc self-test and see if it produces good results?
https://retrocomputing.stackexchange.com/a/14894

GaryM52, 2024-01-10:
I have just run the self-test and I get the System OK message at the end.  There is no disk that will send through to the server. Every attempt is met with the 'crash' screen.  Is there anything in ADTPro that would produce the 'crash screen'? or is it caused by something else.

david-schmidt, 2024-01-11:
All signs are pointing to a hardware fault somewhere before it hits the PC side.  So it may be something in the IIc hardware-wise, or it may be the cleanliness of the drive itself.  The crash screen is the IIc walking off into oblivion.  If every single disk you have does that, then it's more likely to be hardware rather than the software.

It occurs to me that a bad bootstrap (i.e. something wrong with the ADTPro code itself having gone wrong during initial serial bootstrapping) could cause such a problem, but hopefully you created a whole-disk image of ADTPro as mentioned in the instructions (https://adtpro.com/bootstrap.html#Starting_from_bare_metal, step 8) that should minimize the chances of that being the problem.

GaryM52, 2024-01-13:
Thanks for the explanation, I've been thinking my null modem cable may be causing the crash.  I followed the pinout instructions on your web page, but my soldering technique could use some improvement.  It could be the initial connection was OK for the bootstrap and file exchange, but then something may have gone wrong.  I will look at that next.  Thanks again!

GaryM52, 2024-01-17:
I have reworked the null modem cable and am now able to transfer disks to the PC.  Thanks for your help in locating the problem!

david-schmidt, 2024-01-17:
Great, glad that worked out for you!  Enjoy.

closed 2024-01-17 by david-schmidt