ADTPro/adtpro #187: Apple III - ADTPro 2.1.0 - Win 10 - USB 2 RS232

Zeph6969, 2024-04-25:
Hi ,

My issue is that when I try to boostrapping, my Apple III received code and stop on ADTPro "OK" then nothing more (see pict)
I tried many RS232 configurations but same issue, ram (256k) test on the Apple III is ok
End on the trace file is:
"
25/04/24 12:13:41 SerialTransport.pushBuffer() exit.
25/04/24 12:13:41 SerialTransport.pushBuffer() entry, pushing 0 bytes.
25/04/24 12:13:41 SerialTransport.pushBuffer() exit.
25/04/24 12:13:41 Text file sent in 9 seconds.
25/04/24 12:13:41 SerialTransport.getInstructionsDone() getting instructions for: Gui.BS.SOSKERNEL
25/04/24 12:13:41 SerialTransport.getInstructionsDone() returning: 
25/04/24 12:13:41 CommsThread.Worker.run() exit.
"
Do you have a solution?
Thanks for your help :)

[image]

david-schmidt, 2024-04-25:
The evidence on the screen indicates it got through downloading the following files:
1. Kernel
2. ADTPro
3. Drivers

And the final thing it does after sending Drivers is to put up the "/// OK ///" message - meaning it's ready to start executing the OS.  However, you mention in your log that it ended after Kernel (which is only the first of three files it sends).  So either your log is incomplete or... I don't know what else it could be:
https://github.com/ADTPro/adtpro/blob/main/src/client/sos/serial/loader.asm#L141-L146

Anyway, if it halts at "/// OK ///" that means it's not able to progress any further - kicking off the OS is not getting started, which suggests a memory or other processor problem to me.  The code so far is running up in a high part of memory successfully, but SOS exercises every single byte, port, and timer the machine has, and is very sensitive to problems.  A memory test is helpful, but not conclusive.  If you've got the dealer diagnostics disk, that would be a good one to run to see if anything else shows up as problematic.

Zeph6969, 2024-04-25:
Thanks for your answer David.
When I boot SOS from the floppy, the AIII seems to work well. Sadly I don't have the diagnostic floppy disk.
I will try to investigate on bad contacts on ICs, test IC's ram with my tester etc.

david-schmidt, 2024-05-13:
Booting SOS from floppy is a decent stress test, so if that's working and getting you into an app of some sort, that bodes well for your hardware.  What bootable disk(s) do you have?

Zeph6969, 2024-05-14:
Hi David,
Finally I found one bad ram IC and now everything is ok :)
Indeed, as you wrote before, F6E6G RAM test only tests the first 128k, not the entire RAM.

david-schmidt, 2024-05-14:
That's great news!  I'm not sure how it ended up that ADTPro failed where disk-based SOS worked since they are the same code except for a little serial wedge in the middle, as long as your disk had 1.3 on it... but a memory failure will get you one way or another. Congrats on getting that figured out!