ADTPro/adtpro #188: Hello, I do not have a Super Serial Card but I do have a Mountain Computer Multifunction card with a serial port. Will it work with ADTPro?

daerevets, 2024-05-13:


david-schmidt, 2024-05-13:
No, this would require a driver to be written.  The driver would need to take into account the phantom slot setup and hardware addressing  as well as baud rate limitation at 19200.
[Manual link](http://mirrors.apple2.org.za/Apple%20II%20Documentation%20Project/Interface%20Cards/Multifunction%20Cards/Mountain%20Computer%20CPS%20Multifunction%20Card/Manuals/Mountain%20Computer%20CPS%20Multifunction%20Card%20-%20Reference%20Manual.pdf)

closed 2024-05-13 by david-schmidt

daerevets, 2024-05-13:
Thank you for your reply David.

So how is it that the Super Serial Card already has this driver built in? And was it done just for ADTPro? Reviewing the manual I don't see anything about this. For example doing an IN#2 then <ctrl-A>14B, ,<ctrl-A> is confusing to me. This does not seem to be calling a program on the SSC card as I expect it to be a higher address.

Please read on, if  you are interested. I used to work at Capitol Records in Toronto Ontario area back in the early 1980's. Between three of us we created the first computerized system to analyze test tones put on to pre-recorded cassette tapes using  our own designed digital board and analog filter board which plugs into an Apple II plus computer. And this ran for over 10 years doing quality control for the Toronto plant until cassette tape manufacturing was shut down.

I now have that board set (complete with Capital Records asset tag) and am restoring my Apple II plus computer in the interest of getting this analyzer back up and running.

I recreated those test tones on a modern computer and attached it for you to listen to. Depending on your age, it may be a familiar sound to you. I still have all my documentation and marketing stuff on it.

Regards,
Steve

david-schmidt, 2024-05-13:
A very interesting project indeed.

In terms of interaction with serial boards - ADTPro communicates directly at the "hardware" layer of (any of several) boards by interacting directly with the memory-mapped I/O ports of a board.  Each is different, hence the need for a specific driver. 
 The interface you're talking about (i.e. <ctrl-A>, etc.) is known as the "firmware" layer (also known as "Pascal Entry Points") - and it is a standardized layer across many different types of cards that enables compatibility of command and control at the cost of speed.  ADTPro specifically targets this hardware layer in order to achieve that speed, and drivers for some cards including the Super Serial Card have been developed and tested, but not for the Mountain multifunction card.

That being said, I'm reminded that there is a measure of support if you pre-configure your card to have its serial support in slot 2, and select "GENERIC SLOT 2" as your serial device within the ADTPro client - scroll down to "Miscellaneous Serial Devices" on this page:
https://adtpro.com/configserial.html

It's generally less reliable than the card-specific driver and always (far) slower, but it might work with your card.  It hasn't been tested so far.