ADTPro/adtpro #190: Apple IIGS Syntax Error when Bootstrapping ProDOS

ewwitsCooper, 2024-05-14:
https://github.com/ADTPro/adtpro/assets/169733868/2733c085-be49-4102-91c4-e6127bbd9f0c

I used the cables from RetroFloppy.com that were made for ADTPro, I also installed the VCP drivers and the newest version of Java, but when I try sending the ProDOS client, it returns with several syntax errors and it never actually gets to ProDOS.

david-schmidt, 2024-05-14:
It looks like you're really close; the video moves away from what I need to see on the Mac too soon and moves over to the GS too late, so you'll really want to capture the whole thing from both perspectives in order to really tell what's going on.  But I did make a clarifying video a while ago demonstrating some GS funkiness that may help you ensure you're typing all the things you need to over on the GS:
https://www.youtube.com/watch?v=-JYVrLBnh0k

ewwitsCooper, 2024-05-14:
It kept sending ProDOS- No error codes appeared on the Mac

david-schmidt, 2024-05-14:
Ok, then filming the GS screen from the moment it is turned on to the moment you click "ok" to send speediboot (not ProDOS) while commenting on what you are typing will be instructive.

ewwitsCooper, 2024-05-14:
Ill put it up as a new post since I can't post it as a comment

ewwitsCooper, 2024-05-15:
I accepted the default settings on both sides, I've heard multiple people say that

david-schmidt, 2024-05-15:
I can see remnants of the file that is being sent over in clear text, so we know the baud rate is correct.  The first couple of lines are missing, and unfortunately you were panned away right when it started, but later the next few lines appear.  But it's those first couple that give it commands to get into the mode to accept the numbers as code, and that's not working.  It's possible line endings are giving it trouble... it's simply sending a text file over the wire at this stage.

Let's have a look at the serial port configuration on the IIgs.  With the instructions here:
https://adtpro.com/connectionsserial.html#MiniDIN8
...scroll down to where the paragraph starts, "For the IIgs, you will also need to verify a couple of things in the control panel."

Navigate to slot2:
[image]

...and have a look at the setup there.  Does it look the same as on the picture?
[image]

huang-zhen, 2025-01-30:
I have a similar issue on Apple IIe. Some of the machine code sent to the Apple IIe are garbled. During a bootstrap, a typical output on the Apple IIe looks like this:

*7484:73 8E A 3 20 BD 73 6 48 AD
*748E:00 C0 C9 B F0 22 AD 89C0 29
....

(Some bytes of the machine code are received incorrectly on the Apple IIe, as such bytes become one digit or four digits instead of the correct two digits. )

or
]cCall -151
.....
(The first lower case 'c' should've not been received.)

Any suggestion on what might be the cause?