ADTPro/adtpro #191: Can't Open the ADTPro Application - Plus ...

For-ADT, 2024-08-14:
MacOS 10.4.11 & ADTPro 2.0.2.  I can (sorta) use the terminal version of ADTPro but not the version behind the floppy disk icon.  When I click on it the FD icon it shows briefly in the dock and then closes right away.

I am trying to transfer the binary version of the Apple IIGS ROM to use on the Mac with an emulator.  Transferring the ROM from the IIGS I get a .po-suffix result that I can't open on the Mac.  Do I need to suffix the file .bin before I transfer it or ... ?

What all am I missing?

david-schmidt, 2024-08-15:
10.4 is a nice vintage system.  :-)

I think we can tackle the Java startup problem next, but first - this isn't the tool you want to use if you have a single file to transfer.  It's not like Kermit or Zmodem.  You're not transferring files or sets of files... ADTPro transfers whole disk images.  So the ROM from a IIgs isn't going to be a .po file, because it's not a disk image.  You might have put the ROM contents onto a floppy disk and that's what you want to transfer, but that's not what you described...

For-ADT, 2024-08-15:
Hi David - Thanks for your response and thanks also for a very interesting application.

Yeah, I had already figured out that I was way out of it trying to transfer the Apple IIGS ROM file.  Got in too big a hurry and didn't think it thru'.

Likewise I was going with the idea that the JAVA version that comes with OS 10.4 was what I needed.  I have scouted around looking at the required 2.1-7 version and have found I haven't the foggiest idea how to install it.  I am Terminal averse but it seems that some intelligent effort will be required and I need much better clarity of what needs to be done and how to do it than I have now.  Let's start here.

david-schmidt, 2024-08-27:
You'll want to get a little more comfortable with the Terminal app - that's where we're going to be doing much of this work, at least at first.  The first thing you're going to want to do is bring up a terminal and report what is returned from this command (after which you can close it for the time being):
`java -version`

For-ADT, 2024-08-27:
Hi David

I had just returned to my ADTPro project - can you tell when someone is skulking around?

java -version returns --> java version "1.5.0_19" <--

For-ADT, 2024-08-27:
The rest of the return

Java (TM) 2 Run Time Environment, Standard Edition (build 1.5.0_19 b02-306)
Java Hotspot (TM) Client VM (build 1.5.0_19-138, mixed mode, sharing)

david-schmidt, 2024-08-27:
For your vintage of machine, you probably want to go back to version 2.0.3:
https://github.com/ADTPro/adtpro/releases/tag/v_2_0_3
I believe that will likely work better for you.  It doesn't require quite the same high level of Java (which won't exist in your OS combination).  I'd remove whatever you've got downloaded or installed for 2.1.0 and give 2.0.3 a try.

For-ADT, 2024-08-27:
I started with (and only have installed) ADTPro version 2.0.2.  That is the recommended for MacOS 10.4?

david-schmidt, 2024-08-27:
Ok, great - that should work fine.  Let's head back to the terminal, and we need to get into the folder where you unpacked the .dmg.  It's likely in your `/Applications/ADTPro-2.0.2` folder.  So this is what to try in the terminal:
```
$ cd /Applications/ADTPro-2.0.2
$ ls -la    <== this is just for our own edification
$ ./ADTPro.command
```
So the output from `ls -la` will tell us some information about where you're starting from, and the effect of `./ADTPro.command` will show us the output of the system as it tries to bring the application up.

For-ADT, 2024-08-27:
This is really basic - how does one end a line and start a new one in Terminal?

david-schmidt, 2024-08-27:
Hit the "return" key - no need for the `$`, that's the typical prompt that will be presented to you on a new line

For-ADT, 2024-08-27:
This is where I'm screwed up.  The return key brings the response
-bash : $: command not found

For-ADT, 2024-08-27:
Had an extra $ in the line.  Without it the return is
-bash: $: No such file or directory

david-schmidt, 2024-08-27:
Here's what a terminal interaction looks like on my system (with 2.1.0 instead of 2.0.2):
[image]

For-ADT, 2024-08-27:
OK. While you were away I psyched out that my Application folder was not being addressed right. Having worked around this to ADTPro-2.0.2/ADTPro.command , Terminal output shows

ADTPro Server version 2.0.2

Experimental:  JNI_Onload called

Stable Library
==============================

Native lib version =  RXTX 2.1-7
Java lib version    =  RTXT 2.1-7
args length = 0

It also starts up the application, bringing up the window with the connection options

I can't seem to make the other commands work

For-ADT, 2024-08-27:
BTW I'm communicating on my regular desktop - the ancient PowerBook with OS 10.4 on it and the communicator to my Apple IIGS sits on a workbench nearby.

For-ADT, 2024-08-27:
Oops, past my wife's patience limit.  

Maybe we can tackle the java mystery tomorrow.

Thanks so much for your time and effort.  We will gitter dun!

david-schmidt, 2024-08-28:
So it looks like you're in good shape as far as kicking off the application goes.  Next, you're going to want to figure out what USB-serial adapter your PowerBook has on hand and is able to connect with.  You should follow along in the documentation here:
https://adtpro.com/connectionsserial.html#USB_cabling

For-ADT, 2024-08-28:
Good morning David

I have a known-good Keyspan adapter with the proper software and a newly
purchased "crossover" serial cable.

david-schmidt, 2024-08-28:
Great, let us know how you get on with that.  Sounds like you're good to go.

For-ADT, 2024-08-28:
I don't need to do anything with Java?

david-schmidt, 2024-08-28:
You said the app came up (your initial problem description).  We got it up and running - now, does it work?

For-ADT, 2024-08-28:
Well, no.  The ADTPro behind the floppy disk icon still doesn't open.  It flashes briefly in the dock and closes again.  The terminal version - ADTPro.command - opens but doesn't transfer anything.

david-schmidt, 2024-08-28:
I'm going to need (much) more detail about how the command version works (or doesn't).

For-ADT, 2024-08-28:
The server end seems to be OK.
[image]
[image]

On the IIGS

IN#2
<ctrl-A>14B    or
IN#2
<ctrl-I>          gives        
?SYNTAX ERROR

For-ADT, 2024-08-28:
Oops
On the IIGS
IN#2
<ctrl-A>14B
or
IN#2
<ctrl-I>
gives
?SYNTAX ERROR

For-ADT, 2024-08-28:
For some reason the ctrl between the <> doesn't stay in the text?

david-schmidt, 2024-08-28:
What the instructions are trying to say is you hold down the key marked `control` (I don't have a GS in front of me now, maybe it's `ctrl`), and while that key is held down, press then release the key marked `A`, then release the control key.  That's what `<ctrl-A>` means.

david-schmidt, 2024-08-28:
It would be good to watch the video on the bootstrapping page - while the screens have changed a bit since that video was made, the keyboard manipulations you do on the Apple II side have not changed:
https://adtpro.com/bootstrap.html

For-ADT, 2024-08-28:
Oh, good grief.  I've only been around computers since 1988.

For-ADT, 2024-08-28:
Thank you for your patience, guidance and a fantastic piece of software!
[image]