ADTPro/adtpro #192: ADTPro and 40 track 5-1/4" Floppies

Mr-Fixit58, 2024-08-17:
Hi

I have an Apple //e enhanced and I have been using ADTpro to move floppy images and I have some that are 40 tracks. I was using a Quentin floppy drive to make them, and they worked well, but I would like to know if ADTpro can handle the extra tracks when making a disk image
Is there a code change I can make to allow this to happen?

thanks

david-schmidt, 2024-08-18:
It kind of depends on how the driver works for this 40-track drive.  Mostly ADTPro checks if the hardware is compatible with a Disk II, and if so, it takes shortcuts and has special code to deal with 35 tracks much faster than a plain ProDOS block device can.  If it can be convinced that the driver is a block interface to something more generic, it wouldn't matter if it had 40 tracks or 4000; the normal ProDOS interface would communicate how many were there and normal communications could take place for exactly that many.

david-schmidt, 2024-08-30:
I'm not finding anything about a ProDOS driver for this disk.  Unless one exists or one is written, it's unlikely this is a scenario that would work.  The 35 track geometry is baked pretty deeply into ADTPro's logic and memory/buffer layout (as well as ADT's before it); it would be a pretty substantial rewrite to recreate what is essentially the driver itself inside ADTPro for this drive.

closed 2024-08-30 by david-schmidt