ADTPro/adtpro #193: No communication between ADTPro 'host' and Apple IIe 'client'

sbaker51, 2024-10-15:
I have tried to get ADTPro version 2.1.0 to work using a ‘straight through’ DB25 cable using the super serial card configured as ‘terminal’ and have tried to get it to work using the same cable with a ‘null modem’ adapter on it using the super serial card configured as ‘modem’. 

In neither case does the platinum apple IIe show any data being transferred. 

The ‘straight through’ cable was tested with a Fluke 77 VOM using its continuity function to verify that each pin is connected to the corresponding pin number on both DB25 connectors of said cable. 

The Apple IIe has version 1.2.1 on its SCSI hard disk. I wondered if there was a problem using ADTPro version 2.1.0 on the ‘host’ side with version 1.2.1 on the ‘client’ (Apple II) side, so I downloaded version 1.2.1 from your github repository, but it made no difference. 
I understand that to ‘bootstrap’ an Apple that it is necessary to boot it without an operating system. This was also tried to no effect. 
I’ve also tried ‘bootstrapping’ the Apple using version 2.1.0 on an Ubuntu Linux machine again without success. 
In addition, I also tried trying to connect on Windows 10 and Ubuntu using the cassette inputs. 

I used the cassette ‘in’ port (just to the left of the ‘game’ port as seen looking at the Apple from the rear). The PC side of this connection was plugged into the ‘green’ line1/headphone output port on a SoundBlaster Audigy sound card. This was necessary because ASUS offers no modern drivers for the audio chip on the ASUS  P5B-PLUS motherboard. 

Again, the audio cable was checked using the Fluke 77 VOM to verify that the tip and ring of the 1/8 inch audio cable connectors are good. 
 
In an attempt to eliminate the Audigy sound card as a potential issue, I connected the above-mentioned audio cable to an Intel motherboard’s (the one running Ubuntu) line out/headphone port again without success. 

I tested audio volume levels at 10,20,30,40,50,60,70,80,90 and 100% on both the Windows 10 machine (using the Audigy card port) and with the Ubuntu machine’s motherboard line out/headphone ports again without success. 
Note: the switch 1 and switch 2 switches on the super serial card are both set as specified in your instructions SW1: 1001111 SW2: 1101100

I have no idea what I might be doing incorrectly, so any advice would be appreciated. 

[PC Configuration Data.docx](https://github.com/user-attachments/files/17380912/PC.Configuration.Data.docx)

sbaker51@ptd.net

david-schmidt, 2024-10-15:
Interesting that there's an old 1.x client on the IIe.  Was that you that put it there and used it, or is this a "new to you" Apple IIe?

sbaker51, 2024-10-15:
Stanley Baker

 

118 Laufer Ln. 

Apt. 1

Milford, PA 18337

 

(Email)  ***@***.***

david-schmidt, 2024-10-17:
Let me know if you're still there - I've got a lot of clarifying questions, but I've learned not to ask them all at once.  Nothing came through on your last email response... it's better to use GitHub directly rather than email.

sbaker51, 2024-10-18:
David,

 

Thank you for your response. I did send a great deal of information via email, but despite using a Google Drive link and embedding the rest into an MS Word document, it seems the email server rejected it probably because it exceeded the email server’s message size limitation. 

 

I will try to send the same thing via github with the hope that it does not have the same response limitations. 

 

 

Stanley Baker

 

118 Laufer Ln. 

Apt. 1

Milford, PA 18337

 

(Email)  ***@***.***

sbaker51, 2024-10-18:
I downloaded it and installed it perhaps 15 years ago via a serial interface using a different Intel motherboard probably running Windows 7 at the time, so I am completely confused as to why this different (ASUS) motherboard seems to be having issues doing exactly the same thing. 

I have tried getting this to work at least 50 times over the last several days, but in every case, I never see any hex bytes being transferred on the Apple IIe.

Are you aware of any way to test the Windows 10 serial interface to confirm that it is actually sending data over the ‘straight through’ serial interface cable though I have to wonder what efficacy that might have given that Windows 10 clearly shows that it recognizes the serial port on the PC and does not indicate it having any issues at all. 

Powering on the Apple, pressing ctrl-reset to get to an Applesoft prompt ( ] ) followed by typing ‘IN#2’ (because the SSC is installed in slot #2) (I have included a couple of images illustrating the presence of a ‘null modem’ adaptor, as well as illustrating that the SSC is configured as ‘modem’ and that switches 1 and 2 on the SSC are set as specified in your documentation.

 

I have confirmed that the working directory is as it should be
 

Selecting ‘Serial’ from the host interface dialog of ADTPro version 1.2.1 immediately terminates the host interface dialog box. 

Conversely, selecting ‘Serial’ from the host interface dialog of ADTPro version 2.1.0, as you are likely aware only offers two connection speeds – 19200 and 115200.

So to utilize ADTPro version 2.1.0, I changed the serial port speed to 11520 on the PC via the Windows Device Manager…

 

 

Looking at the PC ADTPro host interface shows the following…

 

Clicking ‘Serial’ now shows the following…

 

Selecting ‘Bootstrapping’/’ProDOS’/’SendADTPro Serial Client’ in the PC host interface dialog opens, as expected the ‘ADTPro’ dialog specifying the commands to be typed on the Apple IIe

 

Clicking the ‘Ok’ button results in…

https://drive.google.com/file/d/1YSPijFPNQsQ2r7QNcWOEWUvCUjDjvZJP/view?usp=sharing

The above linked video is 1:34 long, but my iPhone stored it as a 128MB file, thus the need for the Google Drive link. 

As you can see by viewing the video, absolutely nothing happens on the Apple IIe despite following all the directions.

Suggestions?

And thank you very much for you very timely response, it is deeply appreciated.

[image]
[image]

https://drive.google.com/file/d/1xj7p7nSSYDCEluyS3UqT0nBAGLOj7IWu/view?usp=sharing