ADTPro/adtpro #201: ADTPro doesn't work and can't bootstrap: some machine code aren't sent correctly from Ubuntu to Apple IIe

huang-zhen, 2025-01-29:
I have a null-modem connection between a Ubuntu 20.04 and an Apple IIe with APX super serial card. With a baud rate of 19,200, I can run minicom on Ubuntu to communicate with the Apple IIe without any problem. any characters that I typed in minicom on Ubuntu is correctly sent to the Apple IIe, and got displayed and executed. However, I can't get ADTPro to work between the Ubuntu and Apple IIe. I tried many different versions of ADTPro (e.g. 2.1.0, 2.0.3, 2.0.2, 2.0.0, 1.3.0, 1.2.8, 1.2.0, 1.1.9) to do bootstrap and got varying results. A typical output on the Apple IIe looks like this:

*7484:73 8E A   3  20 BD 73 6 48 AD
*748E:00 C0 C9 B F0 22  AD 89C0 29
....

(Some bytes of the machine code are received incorrectly on the Apple IIe, as such bytes become one digit or four digits instead of the correct two digits. )

or 
]cCall -151
.....
(The first lower case 'c' should've not been received.)

I tried to reduce the bootstrap baud rate from 9,600 to 2,400 without any improvement. Is this a software issue with ADTPro? 

I also tried the ADTPro 2.1.0 client on the Apple IIe but it didn't work as well. There is always a "Host timeout" error for any receive, dir, etc. operation.

david-schmidt, 2025-01-31:
You definitely don't want to run at 19200.  You should be at 9600 for bootstrapping.  You need to follow these instructions more or less exactly... basically, don't change any of the defaults even if you're tempted:
https://adtpro.com/bootstrap.html#Starting_from_bare_metal

huang-zhen, 2025-02-01:
I didn't change the baud rate to 19,200 for bootstraping. I used the default 9,600 baud rate for bootstraping and 19,200 baud rate for regular transmitting.

david-schmidt, 2025-02-01:
Did bootstrapping at 9600 work, or is it this 9600 rate that is not working for you?  If bootstrapping isn't working, you should try increasing the pacing time to see if that helps.  A stock Apple II that is functionally fit should have no trouble keeping up, though.  Is it possible you already have an OS booted when you're trying this?