ADTPro/adtpro #206: Bootstrapping greyed out

jimclake, 2025-05-01:
Trying to connect 64-bit Windows 10 laptop to Apple IIGS but Bootstrapping option at top of menu is greyed out.

Running a null modem cable to a serial-USB adapter and into USB port on the laptop.

java -version on my laptop gives me this:

C:\Users\Administrator\Downloads\ADTPro-2.0.2\ADTPro-2.0.2>java -version
openjdk version "11.0.16.1" 2022-08-12 LTS
OpenJDK Runtime Environment Microsoft-40648 (build 11.0.16.1+1-LTS)
OpenJDK 64-Bit Server VM Microsoft-40648 (build 11.0.16.1+1-LTS, mixed mode)

Do I need to be running 32-bit Java?

Thank you.

david-schmidt, 2025-05-01:
There's a couple of things that give me pause here.  I'm not sure how well serial libraries play with OpenJDK, but that's just because I've not used it before.  Assuming that's actually OK, then you need to be absolutely sure the bit width of both your host OS and your Java version match.  If one is 64-bit, they both must be.  If one is 32, then both must be.

Finally, 2.0.2 is pretty old, and it uses a deprecated Java native serial library (rxtx) that was replaced by a newer one a few years ago (jssc) because of aging Java issues.  I'm not sure 2.0.2 is going to work on that new of a JRE in any case.

jimclake, 2025-05-02:
Thanks, David! Upgrading to the 2.1 version did the trick!